coolnameismy/BabyBluetooth #4: 蓝牙扫描

binghuiwb, 2015-09-24:
未扫描成功应建一个方法，用以其他处理

coolnameismy, 2015-09-25:
已经添加6个相关委托，通过以下方法调用时进入委托

````objc
    [baby cancelScan];
    [baby cancelPeripheralConnection:peripherals];
    [baby cancelAllPeripheralsConnection];
````
设置委托的方法(支持AtChannel的调用方式)：

````objc
  //设置断开全部外设连接的委托
    [baby setBlockOnCancelAllPeripheralsConnectionBlock:^(CBCentralManager *centralManager) {
        NSLog(@"setBlockOnCancelAllPeripheralsConnectionBlock");
    }];
    //设置断开指定外设的委托
    [baby setBlockOnCancelPeripheralConnectionBlock:^(CBCentralManager *centralManager, CBPeripheral *peripheral) {
        NSLog(@"setBlockOnCancelPeripheralConnectionBlock");
    }];
    //设置取消扫描的委托
    [baby setBlockOnCancelScanBlock:^(CBCentralManager *centralManager) {
        NSLog(@"setBlockOnCancelScanBlock");
    }];
````

closed 2015-09-25 by coolnameismy

ShawEleven, 2017-08-14:
可以添加设置扫描时长的方法么？