coolnameismy/BabyBluetooth #7: 在iOS上，连接了一个CBPeripheral之后会自动连接其他的CBPeripheral

yangchenghu, 2015-11-01:
LZ你好
最近在使用您的这个框架来调试蓝牙
我遇到了一个问题
我主动链接一个CBPeripheral之后，读取了该设备的Services和characteristics之后，会自动连接之前扫描到的其它的CBPeripheral。不知道这个该如何修改
感谢感谢

coolnameismy, 2015-11-01:
对于连接设备的行为可以通过一个fliter控制。设置连接设备的规则

发自我的 iPhone

coolnameismy, 2015-11-01:
可以通过设置连接设备的fliter，去控制需要连接设备的规则。下面演示一个连接第一个AAA开头的设备
````swift
    /*
    *scan for then connect：1:setting filter 2：scan and connect
    */
    //1:setting filter
     __block BOOL isFirst = YES;
    [baby setFilterOnConnetToPeripherals:^BOOL(NSString *peripheralName) {
        //rule：first peripheral and name start with "AAA"
        if(isFirst && [peripheralName hasPrefix:@"AAA"]){
            isFirst = NO;
            return YES;
        }
        return NO;
    }];
````

closed 2015-11-01 by coolnameismy

yangchenghu, 2015-11-01:
哦哦，明白了，谢谢您

yangchenghu, 2015-11-02:
LZ你好，我设置了setFilterOnConnetToPeripherals 这个回调方法，下了断点之后一直没有被调用，麻烦帮忙讲解一下，谢谢谢谢

coolnameismy, 2015-11-02:
你必须要执行connectToPeripherals（）方法后，才会在扫到设备后进入filter问询是否连接设备，示例如下：
````swift
    //2 扫描、连接
    baby.scanForPeripherals().connectToPeripherals().begin()
````

你可以看wiki里面，有更多的使用说明

yangchenghu, 2015-11-02:
嗯嗯，我已经进行了connect之后没有捕捉到那个断点，我用`setFilterOnConnetToPeripheralsAtChannel: filter:`这个方法限制了自动链接其它的Peripherals，谢谢楼主讲解，感谢提供这么好用的框架 ：）

flymorn, 2017-12-31:
你好，执行 

`baby.having(myperipheral).connectToPeripherals().begin();`

直连某个设备后，可以成功连接，但是之后会连接上其他扫描到的设备；
我知道设置连接过滤器可以解决部分问题：

`[baby setFilterOnConnectToPeripherals: ^BOOL(NSString *peripheralName, NSDictionary *advertisementData, NSNumber *RSSI) {}`

但是这个peripheralName参数只能过滤设备的名称，如果**有2个以上的蓝牙设备名称相同**，比如都是AAA，那么我要怎么样才能连接到其中的一个AAA设备呢？

请指教，谢谢。

JohnsonCui11, 2018-03-20:
每个设备有不同的UUID