coolnameismy/BabyBluetooth #9: 在swift中搜索不到设备？？

miws, 2015-11-16:
baby = BabyBluetooth.shareBabyBluetooth()
        babyDelegate()
        baby?.scanForPeripherals()
        baby?.begin()

并且不能象例子中的 baby?.scanForPeripherals().begin()这样使用。
以上代码在控制台只显示了：
>>>CBCentralManagerStatePoweredOn
就没有了

coolnameismy, 2015-11-16:
sorry，babybluetooth暂不支持swift版本。后续会开发swift版本。预计3个月左右。

发自我的 iPhone

miws, 2015-11-16:
我使用混合编程方法已经集成进入工程了，可以调用，只是没有效果。

coolnameismy, 2015-11-16:
嗯。因为objc的链式方程是用block实现的。swift无法用。你执行单个方法可能会进委托，多个一起执行就不行了。

发自我的 iPhone

lyluoyuan, 2015-11-19:
请教个问题，我现在做蓝牙，用bluetooth这个框架，只能搜到和连上苹果设备(mac , iphone)，即使用您这个demo也是这样的，其他有蓝牙4.0的设备搜不到或者连不上

coolnameismy, 2015-11-19:
嗯。因为objc的链式方程是用block实现的。swift无法用。你执行单个方法可能会进委托，多个一起执行就不行了

gnastnosaj, 2015-11-26:
babyBluetooth?.scanForPeripherals()().begin()()
这样写没用吗

gnastnosaj, 2015-11-26:
>>>CBCentralManagerStatePoweredOn
discover perpheral Optional("L28T_00199")
discover perpheral Optional("L28T_00199")
discover perpheral Optional("L28T_00199")
discover perpheral Optional("L28T_00199")
discover perpheral Optional("AirIII")
discover perpheral Optional("L28T_00199")
discover perpheral Optional("L28T_00199")
discover perpheral Optional("AirIII")
discover perpheral Optional("AirIII")

closed 2015-12-10 by coolnameismy