coolnameismy/BabyBluetooth #19: 这是什么情况<CBCentralManager: 0x16eef900> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported!

jones-so, 2016-01-20:
<CBCentralManager: 0x16eef900> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported!

jones-so, 2016-01-20:
<CBCentralManager: 0x16eef900> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported!

jones-so, 2016-01-20:
CBCentralManager: has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported!

jones-so, 2016-01-20:
第5次等待CBCentralManager 打开任然失败，请检查你蓝牙使用权限或检查设备问题。

我的蓝牙已经打开了，为什么会有这个错误呢？

coolnameismy, 2016-01-20:
Xcode 版本，iphone调试机器版本是多少

发自我的 iPhone

jones-so, 2016-01-20:
xcode是5.1.1，真机ipad 7.0.4，我去掉了willRestoreState代理方法，错误没有了，但是CBCentralManager 仍然打不开，谢谢

coolnameismy, 2016-01-22:
是不是还有设备硬件的限制，查了苹果网站也没找到，直说要求ios6.0以上，你的设备应该可以。你试试用你的iphone做蓝牙开发吧

yuhanle, 2016-03-01:
has no restore identifier 初始化的参数设置有问题
使用这个方法需要在初始化centerManager的时候，像这样写

```
centralManager          = [[CBCentralManager alloc] initWithDelegate:self queue:dispatch_get_main_queue()
                                                                     options:@{
                                                                               CBCentralManagerOptionRestoreIdentifierKey:@"QWSRestoreIdentifier",
                                                                               CBCentralManagerOptionShowPowerAlertKey : @(YES)}];
```

简单添加一个identifier就没问题了~

closed 2016-03-10 by coolnameismy

lxfwelldone, 2020-12-08:
这个修复了吗？为啥我还是遇到，姿势不对？？？