coolnameismy/BabyBluetooth #23: 常见问题：为什么扫描不到外设？

coolnameismy, 2016-03-02:
## 情况1：设备不支持
-  babybluetooth支持BLE4.0蓝牙设备，如果是2.0的设备，那肯定扫描不到
-  模拟器不支持蓝牙调试，必须使用真机调试
-  ios操作系统至少要在6.0以上

## 情况2：外设没有开启广播
-  默认情况下打开手机或者ipad的蓝牙，在使用babybluetooth是扫描不到设备的，因为手机和ipad只是打开了蓝牙设备的电源而已，并没有发送广播，如果想扫描到设备，需要编程的方式，通过iphone或ipad，mac发送蓝牙广播。demo程序中有一个iphone和一个mac发送蓝牙广播的程序。
-  已连接的设备会停止广播，所以也搜索不到

##  情况3：为什么系统的蓝牙能找到设备，但是使用babybluetooth的demo却无法找到？
这种情况一般是因为那个蓝牙设备是支持MFI的，并不是普通的ble设备。

simplelife-patrick, 2016-03-16:
两台iPhone6 Plus，iOS 9.0.2，一台运行BabyBluetoothAppDemo，另一台运行BabyBluetoothIOSStub，为什么还是扫描不到呢？

jiqiuyucool, 2017-08-29:
找了2个Demo 别的Demo找到了蓝牙设备，这个没有找到那个蓝牙设备，都在旁边，奇怪~~~

Verchen, 2017-10-17:
支持MFI的蓝牙设备babybluetooth就无法搜索到了吗，有什么办法解决吗

brucehelen, 2017-10-23:
@Verchen 遇到同样的问题，MFI和BLE有啥区别，为什么BabyBluetooth扫描不到？

yfming93, 2018-03-31:
@jiqiuyucool 请问是找的那两个demo啊。我找了N多demo都扫描不到我外设蓝牙

PiPiPrawn, 2018-05-25:
支持MFI的蓝牙设备babybluetooth就无法搜索到了吗，有什么办法解决吗