coolnameismy/BabyBluetooth #31: 常见问题：外设断开后的自动重连

coolnameismy, 2016-03-16:
我们常常因为一些未知原因导致设备断开连接后，比如因为距离远，设备断电，设备不稳定等原因断开连接后，需要重新连接才能正常使用。重新连接的方法很简单，只需要在设备断开后的委托中，重新调用一下连接外设的方法即可使连接条件回复后重新连接外设。

## 示例：

````objc
    //断开连接的委托
    [self.baby setBlockOnDisconnect:^(CBCentralManager *central, CBPeripheral *peripheral, NSError *error) {
        NSLog(@"设备：%@--断开连接",peripheral.name);
       [weakSelf.baby.centralManager connectPeripheral:peripheral options:nil];
    }];
 //如果原生委托则在这个委托中执行
- (void)centralManager:(CBCentralManager *)central didDisconnectPeripheral:(CBPeripheral *)peripheral error:(NSError *)error {
}
````

## baby中对重连更简单的支持
>  baby最新版本中增加了对断线重连的支持

````
    //设置重新连接的设备
     [baby AutoReconnect:peripheral];
````
通过设置一个自动重连对象，当这个peripheral断开后，会在baby内部的centralManager自动重新连接它。

当然，你也可以调用 ` [baby AutoReconnectDelete:peripheral]; `方法取消自动重连对象

zqzal, 2016-08-24:
很不错.

OrangesChen, 2017-03-01:
Demo里自动重连，如果不手动滑动tableView的话收不到数据，数据不会更新。

coolnameismy, 2017-03-24:
@OrangesChen  那是ui层的问题，可以自行解决

Mirzhou, 2017-04-16:
为什么我第一次启动App的时候走设置读取characteristics的委托 方法可以得到两个characteristics 一个读写、一个通知，断开重连接以后就只能得到一个characteristic 读写，没有通知characteristic！

Mirzhou, 2017-04-16:
我发现断开连接后要先取消通知，否则重连就这样