coolnameismy/BabyBluetooth #37: 快速连接上次已经配对过的设备

fuchanghuan, 2016-04-10:
1，我曾经已经配对过的设备，能否在下次打开时，不用在去扫描设备，而是快速，直接连接上上呢？
我看文档里面提供类似：

retrievePeripherals...........

方法。

coolnameismy, 2016-04-10:
暂未提供这个方法，会在这几天加进去，敬请留意更新

发自我的 iPhone

beiliubei, 2016-04-12:
什么时候更新这个方法？现在遇到一个问题，就是回链的时候，速度不是很快

Sean-Wang, 2016-04-12:
+1

这个功能在实际的应用过程中很需要啊。希望添加这一功能后 Pod 能有小版本升级，让使用 Pod 管理第三方库的用户能第一时间用上。

fuchanghuan, 2016-04-12:
CBCentralManager *manager = [[CBCentralManager alloc] initWithDelegate:self queue:nil];
    NSUUID *uuid = [[NSUUID alloc] initWithUUIDString:@"UUID"];
    NSArray *array = [manager retrievePeripheralsWithIdentifiers:@[uuid]];

这个方法（retrievePeripheralsWithIdentifiers）我测试过了可以快速拿到指定的外设，不过UUID要用下边的方式去取：

 NSString *uuidString = [NSString stringWithFormat:@"%@", [[peripheral identifier] UUIDString]];

coolnameismy, 2016-04-14:
mark,周末更新，最近比较忙。

coolnameismy, 2016-04-15:
本周末更新

发自我的 iPhone

coolnameismy, 2016-04-17:
@beiliubei @fuchanghuan @Sean-Wang 

新添加了api 
````
/**
 根据外设UUID对应的string获取已配对的外设
 
 通过方法获取外设后可以直接连接外设，跳过扫描过程
 */
- (CBPeripheral *)retrievePeripheralWithUUIDString:(NSString *)UUIDString;
````

一个工具方法，用来根据UUIDString获取外设，晚一些时候会重新打包到pod，目前在最新的代码中可以获取，使用示例：

````objc
//快速获取外设
CBPeripheral *p = [baby retrievePeripheralWithUUIDString:@"B19A6ED7-29D5-67EF-0207-6F5AE8BC337B"];
//直接根据外设连接
baby.having(p).connectToPeripherals().begin();
````

另外，如果不适用最新的api,也可以使用下面的方式达到同样的效果。

````objc
  NSUUID *uuid = [[NSUUID alloc]initWithUUIDString:@"B19A6ED7-29D5-67EF-0207-6F5AE8BC337B"];
   CBPeripheral *p = [self.baby.centralManager retrievePeripheralsWithIdentifiers:@[uuid]][0];
baby.having(p).connectToPeripherals().begin();
````

Sean-Wang, 2016-04-17:
@coolnameismy 谢谢。又新增一个非常棒的功能。等待 Pod 升级后就用上这个功能。

coolnameismy, 2016-04-17:
你现在也可以用的。

发自我的 iPhone

fuchanghuan, 2016-04-17:
@coolnameismy  谢谢！谢谢！

closed 2016-04-17 by fuchanghuan

Sean-Wang, 2016-06-08:
使用直接连接模式失败 ;-( 具体代码如下：

    CBPeripheral *p = [BLE retrievePeripheralWithUUIDString:bleIdString];

    DLog(@"before p == %@", p);

    BLE.having(p).and.channel(@"BLE").then.connectToPeripherals().discoverServices().discoverCharacteristics().readValueForCharacteristic().discoverDescriptorsForCharacteristic().readValueForDescriptors().begin();

    DLog(@"after p == %@", p);

DLog 输入结果：

    [Line 698] before p == <CBPeripheral: 0x1288b7df0, identifier = 2E7EF6F2-3EFD-B8FF-F576-45BDFAEBCCC7, name = YS-北京, state = disconnected>
    >>>已切换到BLE
    [Line 700] after p == <CBPeripheral: 0x1288b7df0, identifier = 2E7EF6F2-3EFD-B8FF-F576-45BDFAEBCCC7, name = YS-北京, state = connecting>

一直得不到包括 setBlockOnConnectedAtChannel 或者 setBlockOnFailToConnectAtChannel 在内任何通知。

另外，所有其他代码不变，但使用手动连接途径正常，代码如下：

        CBPeripheral *connectDevicePeripheral = [peripherals objectAtIndex:indexPath.row];

        DLog(@"before connectDevicePeripheral == %@", connectDevicePeripheral);
        
        BLE.having(connectDevicePeripheral).and.channel(@"BLE").then.connectToPeripherals().discoverServices().discoverCharacteristics().readValueForCharacteristic().discoverDescriptorsForCharacteristic().readValueForDescriptors().begin();

DLog 输入结果：

    [Line 698] before connectDevicePeripheral == <CBPeripheral: 0x1270c81b0, identifier = 2E7EF6F2-3EFD-B8FF-F576-45BDFAEBCCC7, name = YS-北京, state = disconnected>