coolnameismy/BabyBluetooth #47: 程式執行5秒後所有掃描都會停止

sjacs5537, 2016-05-13:
雖然執行以下的scanForPeripherals():
```
[weakBaby setBlockOnDisconnect:^(CBCentralManager *central, CBPeripheral *peripheral, NSError *error) {
        NSLog(@"Peripheral Disconnected");      
        // We're disconnected, so start scanning again
        weakBaby.scanForPeripherals().begin();
    }];
```
但在BabyCentralManager.m 裡面的didDiscoverPeripheral執行了connectTimer:
```
- (void)centralManager:(CBCentralManager *)central didDiscoverPeripheral:(CBPeripheral *)peripheral advertisementData:(NSDictionary *)advertisementData RSSI:(NSNumber *)RSSI {
    ..............
    //处理连接设备
    if (needConnectPeripheral) {
        if ([currChannel filterOnconnectToPeripherals](peripheral.name,advertisementData,RSSI)) {
            [centralManager connectPeripheral:peripheral options:[currChannel babyOptions].connectPeripheralWithOptions];
            //开一个定时器监控连接超时的情况
            connectTimer = [NSTimer scheduledTimerWithTimeInterval:5.0f target:self selector:@selector(disconnect:) userInfo:peripheral repeats:NO];
        }
    }
}
//停止扫描
- (void)disconnect:(id)sender {
    [centralManager stopScan];
}
```

scanForPeripherals().begin();會被強制停止

coolnameismy, 2016-05-13:
这个问题不存在，请检查自己的代码是否关闭代码

发自我的 iPhone

sjacs5537, 2016-05-13:
當我刪掉`connectTimer = [NSTimer scheduledTimerWithTimeInterval:5.0f target:self selector:@selector(disconnect:) userInfo:peripheral repeats:NO];`
程式就會成功不斷 連接>setNotifyValue:YES>發送> cancelNotify>scanForPeripherals().begin(); >連接....

coolnameismy, 2016-05-13:
你是在其他页面有调用stop（）方法才会定时关闭

发自我的 iPhone

sjacs5537, 2016-05-13:
剛剛查過沒有用過stop()
以下是central的 code：
```
- (void)viewDidLoad
{
    [super viewDidLoad];
    
    // And somewhere to store the incoming data
    _data = [[NSMutableData alloc] init];
    
    [self babyDelegate];
}
- (void)viewDidAppear:(BOOL)animated
{
    [super viewDidAppear:animated];
    
    NSLog(@"All Peripherals Connection Closed");
    
    //停止之前的连接
    [self.baby cancelAllPeripheralsConnection];
    
    NSLog(@"Scanning started");
    
    //设置委托后直接可以使用，无需等待CBCentralManagerStatePoweredOn状态。
    self.baby.scanForPeripherals().begin();
}
- (void)babyDelegate {
    
    __weak typeof(self.baby) weakBaby = self.baby;
    
    /** centralManagerDidUpdateState is a required protocol method.
     *  Usually, you'd check for other states to make sure the current device supports LE, is powered on, etc.
     *  In this instance, we're just using it to wait for CBCentralManagerStatePoweredOn, which indicates
     *  the Central is ready to be used.
     */
    [weakBaby setBlockOnCentralManagerDidUpdateState:^(CBCentralManager *central) {
        
        if (central.state == CBCentralManagerStatePoweredOn) {
            
        }
        else {
            
            
        }
    }];
    
    [weakBaby setBlockOnCancelAllPeripheralsConnectionBlock:^(CBCentralManager *centralManager) {
        
        NSLog(@"setBlockOnCancelAllPeripheralsConnectionBlock");
    }];
    
    [weakBaby setBlockOnCancelScanBlock:^(CBCentralManager *centralManager) {
            
        NSLog(@"setBlockOnCancelScanBlock");
    }];
    
    //设置扫描到设备的委托
    [weakBaby setBlockOnDiscoverToPeripherals:^(CBCentralManager *central, CBPeripheral *peripheral, NSDictionary *advertisementData, NSNumber *RSSI) {
        
        NSLog(@"搜索到了设备:%@ at %@",peripheral.name, RSSI);
            
        // Ok, it's in range - have we already seen it?
        if (self.currPeripheral != peripheral) {
            
            // Save a local copy of the peripheral, so CoreBluetooth doesn't get rid of it
            self.currPeripheral = peripheral;
            
            // And connect
            NSLog(@"开始连接设备");
            weakBaby.having(peripheral).then.connectToPeripherals().discoverServices().discoverCharacteristics().begin();
[... 85 lines cut ...]
     - [centralManager connectPeripheral:peripheral options:connectPeripheralWithOptions];
     - [peripheral discoverServices:discoverWithServices];
     - [peripheral discoverCharacteristics:discoverWithCharacteristics forService:service];
     
     该方法支持channel版本:
     [baby setBabyOptionsAtChannel:<#(NSString *)#> scanForPeripheralsWithOptions:<#(NSDictionary *)#> connectPeripheralWithOptions:<#(NSDictionary *)#> scanForPeripheralsWithServices:<#(NSArray *)#> discoverWithServices:<#(NSArray *)#> discoverWithCharacteristics:<#(NSArray *)#>]
     */
    
    //示例:
    //扫描选项->CBCentralManagerScanOptionAllowDuplicatesKey:忽略同一个Peripheral端的多个发现事件被聚合成一个发现事件
    NSDictionary *scanForPeripheralsWithOptions = @{CBCentralManagerScanOptionAllowDuplicatesKey:@YES};
    
    /*连接选项-> 只在非后台模式下才起作用
     CBConnectPeripheralOptionNotifyOnConnectionKey :当应用挂起时，如果有一个连接成功时，如果我们想要系统为指定的peripheral显示一个提示时，就使用这个key值。
     CBConnectPeripheralOptionNotifyOnDisconnectionKey :当应用挂起时，如果连接断开时，如果我们想要系统为指定的peripheral显示一个断开连接的提示时，就使用这个key值。
     CBConnectPeripheralOptionNotifyOnNotificationKey:
     当应用挂起时，使用该key值表示只要接收到给定peripheral端的通知就显示一个提
     */
    NSDictionary *connectOptions = @{CBConnectPeripheralOptionNotifyOnConnectionKey:@YES,
                                     CBConnectPeripheralOptionNotifyOnDisconnectionKey:@YES,
                                     CBConnectPeripheralOptionNotifyOnNotificationKey:@YES};
    //连接设备->
    [weakBaby setBabyOptionsWithScanForPeripheralsWithOptions:scanForPeripheralsWithOptions connectPeripheralWithOptions:connectOptions scanForPeripheralsWithServices:@[[CBUUID UUIDWithString:TRANSFER_SERVICE_UUID]] discoverWithServices:@[[CBUUID UUIDWithString:TRANSFER_SERVICE_UUID]] discoverWithCharacteristics:@[[CBUUID UUIDWithString:TRANSFER_CHARACTERISTIC_UUID]]];
}
```

sjacs5537, 2016-05-13:
我覺得在5秒內一直執行以上的code, 但在BabyCentralManager.m 裡面的didDiscoverPeripheral的connectTimer是在main thread執行
所以在setBlockOnDisconnect的block內執行weakBaby.scanForPeripherals().begin();會在main thread的connectTimer的時間到後取消掉

coolnameismy, 2016-05-30:
你没给出你执行调用的方法

sjacs5537, 2016-06-17:
不太懂
在viewDidAppear裡面執行了`self.baby.scanForPeripherals().begin();`，如搜到peripheral後不是就會自動執行BabyCentralManager.m 裡面的didDiscoverPeripheral嗎？

coolnameismy, 2016-06-23:
你不用关心BabyCentralManager里面如何执行的，你可以用baby 设置你需要的block委托就可以收到消息了。

closed 2017-01-24 by coolnameismy