coolnameismy/BabyBluetooth #55: 关于功耗问题！

swiftoc-dev, 2016-06-07:
在使用过程中，发现以下两个问题，暂时没有找到解决方法：
1、连接10分钟左右，会断开连接，然后自动重连，频率相当高。
2、功耗非常大，每隔30ms左右进行一次通信，造成功耗过大。
以下图片是安卓和iOS测试对比：
[image]
[image]

coolnameismy, 2016-06-07:
断线的时间是固定的吗？你的现象挺奇怪的，蓝牙协议里面有不少比较底层的协议上的规定，但是作为app应用层，没有能仔细研究过。

断线你用的是connect方法没有开代码吧？
android直连用的是mac地址，ios用的是identify，是不是可能是这个原因？

发自我的 iPhone

swiftoc-dev, 2016-06-07:
1.断线的时间不是固定的，时间都在10分钟左右，有时早一些，有时迟一些。不稳定是可以确定的。安卓手机连接，没有出现这种情况。
2.连接成功后，APP只做了读取硬件设备上传的数据，并没有进行其他操作。
3.安卓连接方式我并不清楚，iOS连接是指定服务UUID的，如下图：
[image]
[image]

@coolnameismy 谢谢。

coolnameismy, 2016-06-07:
10分钟掉线一次我觉得是正常的可以接受的，掉线后发一次重连请求

发自我的 iPhone

coolnameismy, 2016-08-27:
我猜的一些可能的原因，你android端和ios端和设备的通信方式不同。比如订阅一个用的indicate,一个notify，写数据一个用的onresponse，一个用的withoutresponse，如果读写数据都很频繁，是有可能产生你说的这些问题的。

ruhuaby, 2016-10-16:
我觉得可能是你的硬件在处理某个任务时时间用的过长，导致蓝牙连接断开的。

coolnameismy, 2017-01-24:
一般和硬件关系会比较大一些

closed 2017-01-24 by coolnameismy