coolnameismy/BabyBluetooth #60: 已经连接的外设为什么不能直接扫描? 要扫描还是要先执行connectToPerphal?

ldhios, 2016-07-11:
我的场景是这样的存储外设uuid string. 在一个列表里. 通过retrievePeripheralsWithIdentifiers 获取外设对象. 遍历外设数组直接连接外设. connect成功的回调也调用了.  点击设备列表, 把连接成功的外设传给push的控制器. 在push的控制器里拿到的外设的state 已经是connected .  调用 self.baby.having(self.currentPeripheral).and.channel(Channel).discoverServices().discoverCharacteristics().begin(); 就报未执行connectToPeripherals()不能执行discoverServices(). 加上connect方法就不会. 我的外设已经connected上了. 再次connected 不是多此一举吗?

coolnameismy, 2016-08-27:
[image]

看0.7.0的一个xing新特性，已经能满足你的需求

coolnameismy, 2016-08-27:
https://github.com/coolnameismy/BabyBluetooth/wiki/%E6%9B%B4%E6%96%B0%E8%AF%B4%E6%98%8E

closed 2016-08-27 by coolnameismy