coolnameismy/BabyBluetooth #68: 本机已连接的蓝牙设备又被重新扫描到

archerLj, 2016-08-31:
大神，碰到一个问题，望相助：
是这样的，我用LightBlue先连接上我的蓝牙设备A，然后打开我自己的app之后还能扫描到A，并且这时候的A的state变成CBPeripheralStateDisconnected了，直接导致LightBlue断开了与A的连接。但是，我用BabyBluetooth的demo却没有这样的问题（不会扫描出本机其他app已经连接的蓝牙设备）。

我的部分代码是这样的：
self.manager = [[CBCentralManager alloc] initWithDelegate:self queue:nil];
[self.manager scanForPeripheralsWithServices:nil options:nil];
-(void)centralManager:(CBCentralManager *)central didDiscoverPeripheral:(CBPeripheral *)peripheral advertisementData:(NSDictionary *)advertisementData RSSI:(NSNumber *)RSSI {
    if (![self.nDevices containsObject:peripheral]) {
        [self.nDevices addObject:peripheral];
        [self.manager connectPeripheral:peripheral options:nil];
    }
}

看了BabyBluetooth的源码，也没发现有什么设置来避免扫描到其他app或自己的app已连接的蓝牙设备。请问大神有没有碰到这种问题，该怎么解决？

coolnameismy, 2016-09-01:
每个外设只能同时被一个中心连接

发自我的 iPhone

archerLj, 2016-09-02:
理论上是这样，我的问题是经常把本机其他app，或者自己的app中其他centralManager已经连接上的外设给扫描到导致那些已经连接上的外设断开连接。其他手机上已经连接的设备是没事的。

coolnameismy, 2016-09-10:
@archerLj  不可能出现的。一个外设给连接上就不会在发广播，也不会被连上。问题关闭

closed 2016-09-10 by coolnameismy