coolnameismy/BabyBluetooth #71: 关于蓝牙后台发送广播不广播Local Name和Service UUID‘s的问题

sontie, 2016-09-06:
首先，这是当前我遇到了一个蓝牙方面很是头疼的问题，并不是针对BabyBluetooth蓝牙库的，如果您知道的话，请给与一些提示和帮助。当前的流程是手机发送蓝牙广播，硬件设备扫描广播数据，使用广播数据里Local Name手动写入的唯一标示来匹配，之所以采取这种模式是因为，硬件和手机的关系是一对多，而蓝牙连接的话，苹果设备一联接，其他苹果设备就扫描不到了，只能一对一，所以没有使用蓝牙连接。但是这样的话会出现一个问题，就是苹果蓝牙后台广播的限制，原本发送广播时Advertisement Data会有两个段分别是Local Name和Service UUID‘s这两个字段，但如果在后台发送广播时，是不发送在这两个字段的。那么，在后台模式下，所能扫描的广播数据有哪些是可以进行自定义的呢？如果后台广播的问题不能解决，能否有其他方案来实现硬件和手机之间一对多的匹配，这个问题麻烦了，谢谢。

coolnameismy, 2016-09-10:
这个我到没试过，你把进入后来后发出的广播的数据包内容抓出来看看，有些什么字段。

实现硬件和手机一对除了外设-中心的方式，如果数据量不大，可以把数据放到外设的广播包里面，用手机去扫描外设的广播包拿数据。

coolnameismy, 2017-01-24:
可以广播的，LocalName和serviceUUIDS都可以有的，但是peipheralName不可以修改。有一定需要注意，蓝牙广播包长度是有字节限制的，所以名字和service不能太长。 localName不超过29个字符应该是可以的

XanderXu, 2017-05-17:
@sontie 如果是iOS设备,官方文档里是这么写的:

支持peripheral后台运行的模式

如果要支持peripheral角色的后台模式，你需要在Info.plist中的增加UIBackgroundModes键并在值中包含bluetooth-peripheral值。这样系统会唤醒应用来处理读写和订阅事件。

蓝牙框架（Core Bluetooth framework）不仅允许你的应用被唤醒来处理读写和订阅请求，还允许你的应用在后台状态下发送广播。但你必须注意后台时广播与前台时广播是不同的。即便如此，你必须注意后台与前台时广播处理的区别。特别是当你的应用需要在后台发送广播。 
1，CBAdvertisementDataLocalNameKey 这个键会被忽略，并且peripheral的local name不会被广播 
2，CBAdvertisementDataServiceUUIDsKey 的值中包含的所有service uuids都会被放到“overflow”区域；只有ios设备显示指明在搜索它时才会搜索到这些值。 
3，如果所有的处于广播状态的应用都在后台，广播频率将降低。

coolnameismy, 2017-05-17:
没错，官网不支持。manufacturedata也不支持

closed 2017-05-17 by coolnameismy

DarkSherlock, 2021-01-28:
> 可以广播的，LocalName和serviceUUIDS都可以有的，但是peipheralName不可以修改。有一定需要注意，蓝牙广播包长度是有字节限制的，所以名字和service不能太长。 localName不超过29个字符应该是可以的

请教一下，iOS蓝牙的localName是不是不能超过29个字符，超过的话 我用Android设备去扫的话就扫不出来，请问是不是有29个字符长度的限制，有相关的资料吗，网上找了很久没找到