coolnameismy/BabyBluetooth #83: 手机蓝牙写数据如何获取回调状态

ggdkay, 2016-12-13:
/**
写Characteristic成功后的block
|  when did write value for characteristic successed 
*/
- (void)setBlockOnDidWriteValueForCharacteristic:(void (^)(CBCharacteristic *characteristic,NSError *error))block;
以上函数，回调空，而且ios10 报错

LiweiDong, 2016-12-26:
@coolnameismy 能否在writeValue后直接设置当次写数据的block？
`[peripheral writeValue:(nullable NSData *)data
 forCharacteristic:(nullable CBCharacteristic  *)characteristic
              type:(CBCharacteristicWriteType)type
         withBlock:^(CBPeripheral *peripheral,CBCharacteristic *characteristic,NSError *error){
         }];`

coolnameismy, 2016-12-26:
没有这个api吧

发自我的 iPhone

LiweiDong, 2016-12-27:
@coolnameismy 现在是没有，能否实现呢？
方便检查每次发送数据后 有没有成功。

按现在的block逻辑 并不能保证是本次发送的回调。

coolnameismy, 2016-12-27:
可以，下个版本考虑加入

发自我的 iPhone

ggdkay, 2016-12-30:
开发板子蓝牙4.0接收手机app的数据情况，app这边发送数据一段时候，就会发送不出去，这个函数做回调很必要哦。另外，调用以下接口，会因为什么情况不能写成功，不能发送数据？
currentConnectedPeripheral.writeValue(data, forCharacteristic: writeCharacter, type: CBCharacteristicWriteType.WithoutResponse)

closed 2017-01-24 by coolnameismy