coolnameismy/BabyBluetooth #87: NSBluetoothPeripheralUsageDescription crash

yarshure, 2017-01-24:
This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSBluetoothPeripheralUsageDescription key with a string value explaining to the user how the app uses this data.

coolnameismy, 2017-01-24:
thank for you information? i will check it out ???

???? iPhone

? 2017?1?24????9:57?yarshure <notifications@github.com<mailto:notifications@github.com>> ???

This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSBluetoothPeripheralUsageDescription key with a string value explaining to the user how the app uses this data.

-

coolnameismy, 2017-01-24:
alreay fidxed ?thank you again

???? iPhone

? 2017?1?24????9:57?yarshure <notifications@github.com<mailto:notifications@github.com>> ???

This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSBluetoothPeripheralUsageDescription key with a string value explaining to the user how the app uses this data.

-

closed 2017-01-24 by coolnameismy