coolnameismy/BabyBluetooth #100: 扫描就连接，判断蓝牙名，结果连接多个设备

fanxiangyang, 2017-03-10:
1.次问题应该给iOS设备版本无关，我已经用iPad，iPhone测试过
2.主要实现功能是，扫描到名字XXX立即连接蓝牙，结果连接成功代理方法，连接了多个能扫描到的设备（是指 我连接硬件，结果，硬件，我手机，电脑，别人电脑都连接）
```
   //设置扫描到设备的委托
    [baby setBlockOnDiscoverToPeripherals:^(CBCentralManager *central, CBPeripheral *peripheral, NSDictionary *advertisementData, NSNumber *RSSI) {
        NSLog(@"搜索到了设备:%@", peripheral.name);
        if ([peripheral.name isEqualToString:@"XXX"]) {
            //本次只执行一次，一开始我考虑到线程锁问题，但是后来，我改用全局变量存，依然有问题
            NSLog(@"连接设备222222222222222");
            currentPeripheral=peripheral;//全局变量蓝牙
            //我测试1秒以上延时可以，
            dispatch_after(dispatch_time(DISPATCH_TIME_NOW, (int64_t)(1 * NSEC_PER_SEC)), dispatch_get_main_queue(), ^{
                //里面连接没有使用传过去的，而是使用全局变量的蓝牙
                [weakSelf loadData:[peripheral copy]];
            });
        }
    }];
```
可以看看我代码里面的注释和说明，如果需要，可以下载demo：https://share.weiyun.com/358c0f14893127a69b7089fb98bb9e97

3.过滤代理，只过滤需要的那个蓝牙，也会出现问题，但是，不开硬件不会连接，过滤名字长度大于0，也出现问题，应该是连接蓝牙瞬间出现问题

coolnameismy, 2017-03-24:
一般智能设备都会做出2类标识，一类是你目标设备的类别，比如AAA,还有一个是每个设备的个体标识，比如mac地址。举例来说，蓝牙体重秤的peripheral.name定位为AAA，那么你扫描到AAA就是目标类别，每个秤把自己mac地址放到advertisementData中，从advertisementData取到个体的mac，那这个可以作为个体的标识。

closed 2017-03-24 by coolnameismy

fanxiangyang, 2017-03-24:
peripheral.name我是依照这么过滤的，没有使用advertisementData和Mac，但是我的电脑，手机，和硬件产品的name都是不一样的，当没有开硬件（需要过滤的）是不会主动连接任何，一旦打开硬件，而且判断名字后，我把peripheral对象全局保存了，但是代理方法里面还是会连接多个蓝牙。比如我的蓝牙name是AAA,过滤也是AAA,手机是BBB,电脑CCC,当扫描到AAA时，我保存AAA的蓝牙对象去连接，结果连接了AAA,BBB,CCC,不是连接几个AAA,

766, 2017-04-28:
过滤条件跟没用啊， 
if (peripheralName.length >1 && [peripheralName containsString:@"aaaa"]) {
            return YES;
        }
        return NO;
我这样过滤的结果不管bbb、ccc、ddd、阿毛阿狗都去尝试连接

trusbe, 2017-08-28:
没有连接单个设备或者指定设备的接口，默认就是连接所有能够连接的设备，好像是这样子的。

SYJshang, 2018-08-01:
是不是没有在连接的时候断开扫描？