coolnameismy/BabyBluetooth #109: Is there any way to change the CentralManager dictionary options

Rathish11, 2017-04-21:


coolnameismy, 2017-04-24:
of course !  

````
//设置蓝牙运行时的参数
-(void)setBabyOptionsWithScanForPeripheralsWithOptions:(NSDictionary *) scanForPeripheralsWithOptions
                          connectPeripheralWithOptions:(NSDictionary *) connectPeripheralWithOptions
                        scanForPeripheralsWithServices:(NSArray *)scanForPeripheralsWithServices
                                  discoverWithServices:(NSArray *)discoverWithServices
                           discoverWithCharacteristics:(NSArray *)discoverWithCharacteristics;

//设置蓝牙运行时的参数 channel版
-(void)setBabyOptionsAtChannel:(NSString *)channel
 scanForPeripheralsWithOptions:(NSDictionary *) scanForPeripheralsWithOptions
  connectPeripheralWithOptions:(NSDictionary *) connectPeripheralWithOptions
scanForPeripheralsWithServices:(NSArray *)scanForPeripheralsWithServices
          discoverWithServices:(NSArray *)discoverWithServices
   discoverWithCharacteristics:(NSArray *)discoverWithCharacteristics;
````

Rathish11, 2017-04-26:
Thank you for ur immediate response. But my question is from BabyCentralManager class. I found there is no way to turn off the bluetooth power on alert while initialisation, coz the dictionary option for CBCentralManager is given as static, right? Is it possible to make the dictionary option to assign it while initialisation.  Kindly check the code below so that u may know where am saying.

#if  __IPHONE_OS_VERSION_MIN_REQUIRED > __IPHONE_6_0
        NSDictionary *options = [NSDictionary dictionaryWithObjectsAndKeys:
                                 //蓝牙power没打开时alert提示框
                                 [NSNumber numberWithBool:YES],CBCentralManagerOptionShowPowerAlertKey,
                                 //重设centralManager恢复的IdentifierKey
                                 @"babyBluetoothRestore",CBCentralManagerOptionRestoreIdentifierKey,
                                 nil];
        
#else
        NSDictionary *options = nil;
#endif
        
        NSArray *backgroundModes = [[[NSBundle mainBundle] infoDictionary]objectForKey:@"UIBackgroundModes"];
        if ([backgroundModes containsObject:@"bluetooth-central"]) {
            //后台模式
            centralManager = [[CBCentralManager alloc]initWithDelegate:self queue:nil options:options];
        }
        else {
            //非后台模式
            centralManager = [[CBCentralManager alloc]initWithDelegate:self queue:nil];
        }

XanderXu, 2017-05-17:
@coolnameismy 蓝牙的状态保存和还原现在没有支持, CBPeripheralManagerOptionRestoreIdentifierKey是写死的babyBluetoothRestore,代理方法里面是空的
- (void)centralManager:(CBCentralManager *)central willRestoreState:(NSDictionary *)dict {
    
}
希望以后继续开发啊
中文版文档里有讲解http://blog.csdn.net/swibyn/article/details/52096646