coolnameismy/BabyBluetooth #115: 关于出现CBCentralManagerStateUnsupported的问题

dzonel, 2017-05-15:
您好，
打印log：
2017-05-15 14:01:35.126650 BabyBluetoothAppDemo[1598:680455] (null)
2017-05-15 14:01:35.181769 BabyBluetoothAppDemo[1598:680455] viewDidLoad
2017-05-15 14:01:35.460185 BabyBluetoothAppDemo[1598:680455] viewDidAppear
2017-05-15 14:01:35.460501 BabyBluetoothAppDemo[1598:680455] >>>已切换到默认频道
2017-05-15 14:01:35.460875 BabyBluetoothAppDemo[1598:680455] >>> 第1次等待CBCentralManager打开
2017-05-15 14:01:35.469307 BabyBluetoothAppDemo[1598:680455] >>>CBCentralManagerStatePoweredOn
2017-05-15 14:01:35.475639 BabyBluetoothAppDemo[1598:680455] >>>CBCentralManagerStateUnsupported
2017-05-15 14:01:35.475898 BabyBluetoothAppDemo[1598:680455] >>>CBPeripheralManagerStatePoweredOn
2017-05-15 14:01:35.476196 BabyBluetoothAppDemo[1598:680455] >>>CBPeripheralManagerStatePoweredOn
2017-05-15 14:01:37.652845 BabyBluetoothAppDemo[1598:680497] >>> 第2次等待CBCentralManager打开
2017-05-15 14:01:39.848072 BabyBluetoothAppDemo[1598:680497] >>> 第3次等待CBCentralManager打开
2017-05-15 14:01:42.048046 BabyBluetoothAppDemo[1598:680497] >>> 第4次等待CBCentralManager打开
2017-05-15 14:01:44.214740 BabyBluetoothAppDemo[1598:680497] >>> 第5次等待CBCentralManager 打开任然失败，请检查你蓝牙使用权限或检查设备问题。

出现这种情况的代码如下：
baby = [BabyBluetooth shareBabyBluetooth];
babyLV = [[BabyBluetooth alloc] init];

是不是说CBCentralManager只能创建一次？

coolnameismy, 2017-05-17:
iphone4s以下，虚拟机都会出现这种情况，因为机器本身没有Bluetooth4.0设备

closed 2017-05-17 by coolnameismy

dzonel, 2017-07-13:
大神，我问的是：“是不是说CBCentralManager只能创建一次？”  虚拟机没有蓝牙设备我是知道的，我用的真机。虚拟机出现不出现我不知道，反正真机是出现了，iphone5以上，系统10.3

xylinuxkernel, 2019-08-21:
频繁的创建释放，会导致CBCentermanager生命周期混乱，可能出现存在多个CBCentermanager对象，结果第二个就失效了。
处理方法：在BabyCentralManager中提供私有API立即释放CBCentermanager，可以解决。