coolnameismy/BabyBluetooth #122: 用alloc创建的第二个baby对象无法实现搜索

ZL335, 2017-05-26:
用alloc创建第一个baby对象可以实现正常的搜索，但是创建的第二个baby对象去搜索的时候，提示“打开任然失败，请检查你蓝牙使用权限或检查设备问题。”请问这是怎么回事，只能允许一个对象实现搜索功能吗

ZL335, 2017-05-26:
在线等大神出现，发现别人也遇到过同样的问题https://forums.developer.apple.com/message/172382#172382

ZL335, 2017-05-26:
babybluetooth对于设备支持问题，用单例模式进行连接时无法收到多设备返回的信息

yinanwang1, 2019-06-27:
查看源码：
`if (poweredOn) {
        self.CENTRAL_MANAGER_INIT_WAIT_TIMES = 0;
        //扫描后连接
        if (self.babyCentralManager->needScanForPeripherals) {
            //开始扫描peripherals
            [self.babyCentralManager scanPeripherals];
        }
        //直接连接
        else {
            if (cachedPeripheral) {
                [self.babyCentralManager connectToPeripheral:cachedPeripheral];
            }
        }
        return;
    }
    //尝试重新等待CBCentralManager打开
    self.CENTRAL_MANAGER_INIT_WAIT_TIMES ++;
    if (self.CENTRAL_MANAGER_INIT_WAIT_TIMES >= KBABY_CENTRAL_MANAGER_INIT_WAIT_TIMES ) {
        BabyLog(@">>> 第%d次等待CBCentralManager 打开任然失败，请检查你蓝牙使用权限或检查设备问题。", self.CENTRAL_MANAGER_INIT_WAIT_TIMES);
        return;
        //[NSException raise:@"CBCentralManager打开异常" format:@"尝试等待打开CBCentralManager5次，但任未能打开"];
    }`
其中的poweredOn一直是NO，也就是centralManager.state的状态没有PoweredOn，所以会有这个提示。