coolnameismy/BabyBluetooth #136: 为什么点击连接后，会一直连接其他设备，

tracyYBY, 2017-08-03:
baby.having(self.currPeripheral).connectToPeripherals().discoverServices().discoverCharacteristics().readValueForCharacteristic().discoverDescriptorsForCharacteristic().readValueForDescriptors().begin();

会调连接成功，但是也会一直连接其他设备，而且收不到didUpdateValueForCharacteristic 的回调

Verchen, 2017-10-21:
我也是，其它的设备也都一块连上了，请问是怎么解决的

1037390459, 2017-11-08:
[baby setFilterOnConnectToPeripherals:^BOOL(NSString *peripheralName, NSDictionary *advertisementData, NSNumber *RSSI) {
        return [weakSelf.curPeripheral.name isEqualToString:peripheralName];
    }];

在过滤器中过滤就行了

SKbutton, 2019-01-16:
@1037390459  不是说过滤掉就可以了，而是你已经指定连哪一个 
```
baby.having(self.currPeripheral).
```
可结果却是连了所有设备，莫名其妙

yinanwang1, 2019-06-27:
看源码判断
`if (poweredOn) {
        self.CENTRAL_MANAGER_INIT_WAIT_TIMES = 0;
        //扫描后连接
        if (self.babyCentralManager->needScanForPeripherals) {
            //开始扫描peripherals
            [self.babyCentralManager scanPeripherals];
        }
        //直接连接
        else {
            if (cachedPeripheral) {
                [self.babyCentralManager connectToPeripheral:cachedPeripheral];
            }
        }
        return;
    }`
因为调用了connectToPeripherals(),所以needScanForPeripherals为YES，那么就开始扫码所有了。