coolnameismy/BabyBluetooth #143: 蓝牙读取或者写入时闪退，怎么样修复？

kakaka6820, 2017-09-08:
2017-09-07 17:02:05.823146 FutureDreams[748:135770] *** Assertion failure in -[CBUUID initWithData:], /BuildRoot/Library/Caches/com.apple.xbs/Sources/CoreBluetooth/CoreBluetooth-351/CoreBluetooth/CBUUID.m:171

2017-09-07 17:02:22.225554 FutureDreams[748:135770] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Data <> does not represent a valid UUID'
*** First throw call stack:
(0x18dcdc1c0 0x18c71455c 0x18dcdc094 0x18e76682c 0x1948b521c 0x1948b5434 0x1948b38ac 0x1948b3034 0x1948afef4 0x1948ab018 0x1948bef58 0x1948b770c 0x1025b125c 0x1025b121c 0x1025beb54 0x1025b4ce4 0x1025b6060 0x18dc89f2c 0x18dc87b18 0x18dbb6048 0x18f639198 0x193b90628 0x193b8b360 0x1001ff9d8 0x18cb985b8)

LincLiu, 2019-01-22:
同问

LincLiu, 2019-03-25:
再同问

LincLiu, 2019-06-10:
> 2017-09-07 17：02：05.823146 FutureDreams [748：135770] ***断言失败 - [CBUUID initWithData：]，/ BuildRoot / Library/Caches / com.apple.xbs / Source / CoreBluetooth / CoreBluetooth-351 / CoreBluetooth / CBUUID.m：171
> 
> 2017-09-07 17：02：22.225554 FutureDreams [748：135770] ***由于未捕获的异常'NSInternalInconsistencyException'而终止应用程序，原因：'Data <>不代表有效的UUID' 
> ***第一次抛出调用堆栈：
> （0x18dcdc1c0 0x18c71455c 0x18dcdc094 0x18e76682c 0x1948b521c 0x1948b5434 0x1948b38ac 0x1948b3034 0x1948afef4 0x1948ab018 0x1948bef58 0x1948b770c 0x1025b125c 0x1025b121c 0x1025beb54 0x1025b4ce4 0x1025b6060 0x18dc89f2c 0x18dc87b18 0x18dbb6048 0x18f639198 0x193b90628 0x193b8b360 0x1001ff9d8 0x18cb985b8）

https://www.jianshu.com/p/36111b88e1e3这篇博客应该能解决你的问题

yinanwang1, 2019-06-27:
更新版本应该已经解决了 添加try catch来捕获异常