coolnameismy/BabyBluetooth #146: iOS11下有些情况下会出现无法发现服务的问题

change2hao, 2017-09-25:
升级到iOS11，有些手机会遇到 `- (void)peripheral:(CBPeripheral *)peripheral didDiscoverServices:(NSError *)error` 不回调的问题，请问谁遇到过？

leisong1125, 2017-09-30:
我遇到过，但是我是在设备异常断开之后，就发现不了服务了

change2hao, 2017-09-30:
@leisong1125 我只在iOS11下发现了这个问题，目前只有reset网络设置或者关机重启才能解决，不知道你有没有更好的办法

leisong1125, 2017-09-30:
我正要回你，我是在升级硬件过程中，强制断电会导致发现不了服务，之后所有同型号的设备都出现这个问题，我尝试了一下重启手机可以解决，目前还在不断测试中

change2hao, 2017-10-23:
已经解决

closed 2017-10-23 by change2hao

leisong1125, 2017-10-23:
怎么 解决的 什么问题造成的

change2hao, 2017-10-23:
@leisong1125 外设的蓝牙固件问题，设置的最小间隔和最大间隔不符合Apple标准，跟App没有关系

leisong1125, 2017-10-23:
谢谢 您的回答 我们这边 要测试 一线 才能知道 是不是一样的问题

XanderXu, 2017-11-03:
@change2hao 请问这个最小间隔,最大间隔的标准是多少? 苹果有没有文档啊?我也遇到这个问题了

change2hao, 2017-11-04:
https://developer.apple.com/library/content/qa/qa1931/_index.html 希望对你有帮助 @XanderXu

XanderXu, 2018-04-04:
@change2hao 感谢了,我的不是这个问题.但是从你的链接中找到了其他有用信息,是外设的蓝牙特征地址在重启后会变动,但又没有实现serviceChange方法导致的,iOS系统级缓存出现问题.
如果外设的服务和特征地址会变动,按蓝牙规范需要在serviceChange中通知外部刷新...
这个问题花了我两个月才解决...