coolnameismy/BabyBluetooth #149: 重连问题

OneWolfgang, 2017-10-08:
我的app在第一次扫描蓝牙设备后保存了设备的UUID, 然后通过retrievePeripheralWithUUIDString获取到设备类，可以连接。当app重新启动后，通过保存的UUID获取到设备类，再连接就有问题，不进入连接成功的回调也进入连接失败的回调。只有扫描连接成功断开后再重新连接才可以，请问这个是什么原因呢？

XanderXu, 2017-11-03:
没有保存外设peri吧,要调用having(peri)
_baby.having(peri).and.channel(channelOnPeropheralView).then.connectToPeripherals().discoverServices().discoverCharacteristics().readValueForCharacteristic().discoverDescriptorsForCharacteristic().readValueForDescriptors().begin();

OneWolfgang, 2017-11-04:
调用了having(peri)。app重新启动后，直接通过保存的UUID获取设备类连接就不进入各种回调 ，只有扫描一次该设备才可以连接成功，然后可以直接连接成功，但是当app退出重新启动后还必须得扫描一次该设备才可以连接成功。

itmarsung, 2018-04-04:
我也遇到类似的问题。请问你解决了么？

yinanwang1, 2019-06-27:
`// 进行判断是否已经连接
    self.peripheral = [self.babyBluetooth retrievePeripheralWithUUIDString:kServiceUUID];

    if (nil != self.peripheral) {
    self.babyBluetooth.having(self.peripheral).and.channel(self.bikeNoStr).then.connectToPeripherals().begin();

        // 设置超时
        [self performSelector:@selector(sendMessageToBleTimeOut) withObject:nil afterDelay:kTimeOut];

        return;
    }`
如果能够获取到就直接连接peripheral。我用这个代码是可以重连成功的。