coolnameismy/BabyBluetooth #155: 蓝牙外设返回的数据会有缓存

chenrengui, 2017-11-17:
请问为什么要对蓝牙外设返回的数据做缓存,怎么清除缓存呢?调用的方法为notify: characteristic: block:

在block回调里面,得到的蓝牙外设返回的数据,会有缓存,这是什么情况???

lz11, 2018-02-27:
同问  连接上会返回一次以前的数据  怎么清除

yinanwang1, 2019-06-27:
我也碰到了这个问题，查看文档没有地方提到有缓存。感觉像是蓝牙设备（外围设备）的缓存。

chenrengui, 2019-06-27:
在发现特征值的代理方法里，你是不是调用了readValue的方法？

| |
crgtpy
邮箱：crgtpy@163.com
|

Signature is customized by Netease Mail Master

在2019年06月27日 15:29，_阿南_ 写道：

我也碰到了这个问题，查看文档没有地方提到有缓存。感觉像是蓝牙设备（外围设备）的缓存。

yinanwang1, 2019-06-27:
`我在- (void)setBlockOnDiscoverCharacteristicsAtChannel:(NSString *)channel
                                            block:(void (^)(CBPeripheral *peripheral,CBService *service,NSError *error))block`这个方法里，实现了
`- (void)notify:(CBPeripheral *)peripheral
characteristic:(CBCharacteristic *)characteristic`这个方法来监听value的变化。

alanFri, 2019-12-04:
> setBlockOnDiscoverCharacteristics

你这么做解决问题了么

chenrengui, 2019-12-04:
我把这个方法注释掉了，就好了

| |
crgtpy
邮箱：crgtpy@163.com
|

Signature is customized by Netease Mail Master

在2019年12月04日 15:06，alanFri 写道：

setBlockOnDiscoverCharacteristics

你这么做解决问题了么

quietStefan, 2021-07-02:
我注释了- (void)setBlockOnDiscoverCharacteristicsAtChannel:(NSString *)channel
                                            block:(void (^)(CBPeripheral *peripheral,CBService *service,NSError *error))block
还是不行，缓存数据还是从setBlockOnDidWriteValueForCharacteristicAtChannel 返回了。。。