coolnameismy/BabyBluetooth #181: 蓝牙后台的一个问题

c1394393468, 2018-05-23:
我阅读了 iOS 的后台模式
和 蓝牙后台模式。但是依旧有以下的问题。

我们现在有一款蓝牙设备的App，应用可以在后台进行蓝牙操作连接。 我们用状态保存和恢复功能来保持App在后台扫描设备。一般情况下都是正常的。直到用户在后台关闭蓝牙。这个时候后台时间十秒内没有连接上设备。App处于挂起状态，这个时候如果其他App向系统申请内存消耗的时候，系统会把App终止。在这种情况下蓝牙再次关闭开启无法激活App并进行正常的操作。

这貌似和 Bluetooth background 描述不一致，当应用在后台被终止时，就凉了。

```
Performing Long-Term Actions in the Background
Some apps may need to use the Core Bluetooth framework to perform long-term actions in the background. As an example, imagine you are developing a home security app for an iOS device that communicates with a door lock (equipped with Bluetooth low energy technology). The app and the lock interact to automatically lock the door when the user leaves home and unlock the door when the user returns—all while the app is in the background. When the user leaves home, the iOS device may eventually become out of range of the lock, causing the connection to the lock to be lost. At this point, the app can simply call the connectPeripheral:options: method of the CBCentralManager class, and because connection requests do not time out, the iOS device will reconnect when the user returns home.

Now imagine that the user is away from home for a few days. If the app is terminated by the system while the user is away, the app will not be able to reconnect to the lock when the user returns home, and the user may not be able to unlock the door. For apps like these, it is critical to be able to continue using Core Bluetooth to perform long-term actions, such as monitoring active and pending connections.
```

c1394393468, 2018-05-26:
系统回收能够理解，关键在于蓝牙重新开启或者连接没有重新激活应用。并且在被系统回收时也收不到Memory Warning 和 Terminat 的消息。这样如果用来实现Apple 文档中 家居开关锁的话是有问题的，程序可能已经在后台被Kill，但用户却无法知道。

ShenYj, 2018-06-04:
毕竟苹果也是为了省电, 要是真的靠蓝牙能实现无限制的后台, 那隐患也不小. 都做成流氓软件了

c1394393468, 2018-06-04:
Bluetooth Background 文档中描述的
https://developer.apple.com/library/content/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW1

如果蓝牙不能唤醒App，那么IOT App就没有太大的意义了，每次必须要通过前台唤醒操作的话。

ShenYj, 2018-06-12:
你发这玩应做蓝牙开发的基本都看过

c1394393468, 2018-06-12:
嗯， 但还是感谢回答问题