coolnameismy/BabyBluetooth #208: 请教一个问题

ldezho, 2019-04-10:
scanForPeripheralsWithServices:options:
方法中的serviceUUIDs和options中的CBCentralManagerScanOptionSolicitedServiceUUIDsKey对应的serviceUUIDs是什么区别？

yinanwang1, 2019-06-26:
- (void)scanForPeripheralsWithServices:(nullable NSArray<CBUUID *> *)serviceUUIDs options:(nullable NSDictionary<NSString *, id> *)options;
这个方法中有解释：
/*!
 *  @method scanForPeripheralsWithServices:options:
 *
 *  @param serviceUUIDs A list of <code>CBUUID</code> objects representing the service(s) to scan for.
 *  @param options      An optional dictionary specifying options for the scan.
 *
 *  @discussion         Starts scanning for peripherals that are advertising any of the services listed in <i>serviceUUIDs</i>. Although strongly discouraged,
 *                      if <i>serviceUUIDs</i> is <i>nil</i> all discovered peripherals will be returned. If the central is already scanning with different
 *                      <i>serviceUUIDs</i> or <i>options</i>, the provided parameters will replace them.
 *                      Applications that have specified the <code>bluetooth-central</code> background mode are allowed to scan while backgrounded, with two
 *                      caveats: the scan must specify one or more service types in <i>serviceUUIDs</i>, and the <code>CBCentralManagerScanOptionAllowDuplicatesKey</code>
 *                      scan option will be ignored.
 *
 *  @see                centralManager:didDiscoverPeripheral:advertisementData:RSSI:
 *  @seealso            CBCentralManagerScanOptionAllowDuplicatesKey
 *	@seealso			CBCentralManagerScanOptionSolicitedServiceUUIDsKey
 *
 */

大概是说会先用“scanForPeripheralsWithServices:options:方法中的serviceUUIDs”进行搜索，如果这个为nil，那么就返回所有。再进行options的条件过滤。

ldezho, 2019-06-27:
没看明白这两个serviceUUIDs的区别

closed 2019-08-27 by ldezho